Simple-Robotics/aligator #105: No module named 'tap'

stephane-caron, 2024-01-02:
## Bug description

```
Traceback (most recent call last):
  File "/home/scaron/src/aligator/examples/cartpole.py", line 11, in <module>
    from utils import create_cartpole, ArgsBase, plot_convergence, get_endpoint_traj
  File "/home/scaron/src/aligator/examples/utils/__init__.py", line 8, in <module>
    import tap
ModuleNotFoundError: No module named 'tap'
```

## Expected behavior

The example, running.

## Reproduction steps

Steps to reproduce the behavior:
1. Create a new conda environment
2. Install from conda as per the readme instruction: ``conda install -c simple-robotics aligator``
3. Go to the examples directory
4. Run the cart-pole example
5. See error

### System

- OS: Ubuntu 22.04
- aligator version: 0.4.0

stephane-caron, 2024-01-02:
NB: the PyPI workaround is `pip install typed-argument-parser`` (not ``tap``, which is another package importing some code with outdated Python syntax).

ManifoldFR, 2024-01-02:
Ah yes, the dependency on `typed-argument-parser` is only in the pyproject.toml, not the actual conda packaging file https://github.com/Simple-Robotics/aligator/blob/main/packaging/conda/aligator-release/meta.yaml

ManifoldFR, 2024-01-23:
Hey @stephane-caron @jorisv , I could make a quick patch adding `typed-argument-parser` to meta.yaml, would that be alright?

jorisv, 2024-01-24:
@ManifoldFR You can add it as a run requirement (like matplotlib)

ManifoldFR, 2024-09-11:
Fixed by #139 back in April

also, why the hell does saying "Closes #xxx" not work properly ? smh github

closed 2024-09-11 by ManifoldFR

jorisv, 2024-09-12:
@ManifoldFR It's only working when the merge is done on the default branch so it's kind of useless for our workflow :(