Simple-Robotics/aligator #116: Pre-commit

jcarpent, 2024-01-12:
Use pre-commit files from other projects (e.g., https://github.com/humanoid-path-planner/hpp-fcl/blob/8bdad1488c1b43d2da27ae2051ab2b1850f2820c/.pre-commit-config.yaml) to automatically update the rules and apply `pre-commit` on PRs.

ManifoldFR, 2024-01-16:
If I understand:
- integrate the pre-commit bot from https://pre-commit.ci/
- add to the config
  ```
  ci:
    autoupdate_branch: devel
  ```

jorisv, 2024-02-12:
Hi @jcarpent,

Are you sure to want the pre-commit bot ? I find it really annoying.

1. It pushs some patches on code under development, you can then have conflicts when you push after
2. It doesn't encourage you to use formatting tools on your machine

I prefer a red CI, or to set [autofix_prs](https://pre-commit.ci/#configuration-autofix_prs) to false. Then user can comment "pre-commit.ci autofix" if he really doesn't want to install pre-commit…

jorisv, 2024-02-13:
Fixed

closed 2024-02-13 by jorisv