Simple-Robotics/aligator #140: pinocchio3 requirement

m8dotpie, 2024-04-17:
## Bug description

Talos walk example requires pinocchio3 while all aligator installations rely on pinocchio2 and I can't find a way to build aligator from sources in the environment with pinocchio3, either built or installed. I have concluded that it relies on p3 based on the usage of `parentJoint` instead of `parent` and `ProximalSettings`

## Expected behavior

Example with talos work should work. 

## Reproduction steps

1. Run talos_walk.py

## Additional context

Either I don't understand your build pipeline, or some details are missing.

### System

- OS: Ubuntu 22.04

ManifoldFR, 2024-04-18:
Hi !

Good catch! By building with Pinocchio 3, you mean using the current preview branch?

m8dotpie, 2024-04-18:
> Hi !
> 
> Good catch! By building with Pinocchio 3, you mean using the current preview branch?

Yes, as far as I know, only preview branch is avaiable publicly for now

ManifoldFR, 2024-04-18:
> > Hi !
> > Good catch! By building with Pinocchio 3, you mean using the current preview branch?
> 
> Yes, as far as I know, only preview branch is avaiable publicly for now for building from sources (and olivier.roussel::pinocchio for installation from conda)

Oh yes, but unfortunately it's horribly out of date. There's a [recent build on this channel](https://anaconda.org/agm-ws-2023/repo) but it doesn't come with headers that you can use to build aligator against.

I don't think we have a more recent version of Pinocchio 3 with headers available - some of the code contains stuff that's just not public yet (not published as a paper), but we're gearing up for a full public release soon we hope. Until then this example wouldn't actually work, sorry...

m8dotpie, 2024-04-18:
> Oh yes, but unfortunately it's horribly out of date. There's a [recent build on this channel](https://anaconda.org/agm-ws-2023/repo) but it doesn't come with headers that you can use to build [...]

Got it, thank you for clarification, looking forward to your publication!

closed 2024-04-18 by ManifoldFR