Simple-Robotics/aligator #169: WrenchConeResidual contact_id

m8dotpie, 2024-07-10:
WrenchConeResidual (as well as MultibodyWrenchConeResidual) has contact_id argument and it is not quite clear what it represents. 

It is possible to specify arbitrarily large numbers as well as negative IDs and it will work formulate the problem just fine. Is it intended behaviour? I suggest to catch impossible values and/or make it more clear what to specify there.

Distinction between two metntioned residuals (WrenchCone and MultibodyWrenchCone) is not clear from the implementation and arises from the same contact_id uncertainty. Would you be so kind as to clarify?

edantec, 2024-07-16:
Thank you for pointing this inconsistency.

The API for contacts is currently undergoing some important changes, and we intend to fix the issue you mention in the coming release. Right now, contact_id makes reference to the id pointing to the contact model located inside the corresponding std::vector of contacts. We plan to use contact_name instead, allowing us to catch any inconsistency in the provided argument.

WrenchCone and MultibodyWrenchCone implement the same cone residual cost, but one is tailored for centroidal formulation while the other is tailored for full dynamics (or multibody dynamics) formulation.

edantec, 2024-09-30:
The API for force-related costs has been updated and is now using the name of the contact frame as an argument instead of contact ID.

closed 2024-09-30 by edantec