Simple-Robotics/aligator #178: Using SolverFDDP in examples/talos_walk.py

Cfather, 2024-07-24:
## Bug description

Hi, I was trying SolverFDDP in `examples/talos_walk.py`.
But there were some warnings and I'm not sure if this is a valid example for SovlerFDDP:
```
[SolverFDDP] setup: problem stages [0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152, 153, 154, 155, 156, 157, 158, 159, 160, 161, 162, 163, 164, 165, 166, 167, 168, 169, 170, 171, 172, 173, 174, 175, 176, 177, 178, 179, 180, 181, 182, 183, 184, 185, 186, 187, 188, 189, 190, 191, 192, 193, 194, 195, 196, 197, 198, 199, 200, 201, 202, 203, 204, 205, 206, 207, 208, 209, 210, 211, 212, 213, 214, 215, 216, 217, 218, 219] have constraints, which this solver cannot handle. Please use a penalized cost formulation.
```

Here's the verbose output from the solver:
```
iter ｜  alpha   ｜ prim_err ｜ dual_err ｜   preg   ｜   dphi0   ｜   merit   ｜    ΔM     
    1｜5.0000e-01｜6.4917e-01｜3.2403e+02｜1.0000e-09｜-7.7583e+04｜ 1.4379e+04｜-2.5623e+04
    2｜1.0000e+00｜3.2459e-01｜4.4516e+02｜1.0000e-09｜-2.4953e+04｜ 8.4791e+03｜-5.8996e+03
    3｜1.0000e+00｜1.7347e-18｜7.1007e+02｜1.0000e-06｜-1.0581e+04｜ 3.3940e+03｜-5.0851e+03
    4｜1.0000e+00｜1.7347e-18｜1.5924e+02｜1.0000e-06｜-3.2192e+02｜ 3.2379e+03｜-1.5609e+02
    5｜1.0000e+00｜8.6736e-19｜8.3876e+00｜1.0000e-06｜-6.1340e+00｜ 3.2363e+03｜-1.6341e+00
    6｜1.0000e+00｜1.7347e-18｜1.4574e+00｜1.0000e-06｜-8.3854e-01｜ 3.2360e+03｜-3.1077e-01
    7｜1.0000e+00｜8.6736e-19｜1.7334e-01｜1.0000e-06｜-1.6462e-01｜ 3.2360e+03｜-9.3027e-03
    8｜1.0000e+00｜8.6736e-19｜1.0387e-01｜1.0000e-06｜-3.2932e-02｜ 3.2360e+03｜-2.5500e-02
    9｜1.9531e-03｜8.6736e-19｜4.3905e-02｜1.0000e-06｜-6.7444e-03｜ 3.2360e+03｜ 4.6571e-06
   10｜1.9531e-03｜1.7347e-18｜4.3649e-02｜1.0000e-05｜-6.6514e-03｜ 3.2360e+03｜ 4.7193e-06
   11｜1.9531e-03｜1.7347e-18｜4.3395e-02｜1.0000e-04｜-6.4164e-03｜ 3.2360e+03｜ 4.2661e-06
   12｜1.9531e-03｜1.7347e-18｜4.3144e-02｜1.0000e-03｜-5.1490e-03｜ 3.2360e+03｜ 9.6685e-07
   13｜2.5000e-01｜1.7347e-18｜4.2908e-02｜1.0000e-02｜-2.1066e-03｜ 3.2360e+03｜-7.9907e-05
   14｜1.9531e-03｜1.7347e-18｜3.0510e-02｜1.0000e-02｜-1.5275e-03｜ 3.2360e+03｜-1.1920e-07
   15｜1.9531e-03｜1.7347e-18｜3.0367e-02｜1.0000e-01｜-4.0298e-04｜ 3.2360e+03｜ 2.4429e-07
   16｜1.9531e-03｜1.7347e-18｜3.0241e-02｜1.0000e+00｜-1.7968e-04｜ 3.2360e+03｜-2.5193e-08
   17｜1.0000e+00｜1.7347e-18｜3.0119e-02｜1.0000e+01｜-1.1711e-04｜ 3.2360e+03｜-1.3658e-05
   18｜1.9531e-03｜1.7347e-18｜8.1468e-04｜1.0000e+00｜-4.6778e-05｜ 3.2360e+03｜ 3.0587e-07
   19｜1.9531e-03｜1.7347e-18｜1.3245e-03｜1.0000e+01｜-1.0509e-05｜ 3.2360e+03｜ 3.6758e-08
   20｜1.0000e+00｜1.7347e-18｜1.8837e-03｜1.0000e+02｜-1.9503e-06｜ 3.2360e+03｜-9.9702e-07
   21｜1.9531e-03｜8.6736e-19｜1.2362e-03｜1.0000e+01｜-8.7057e-06｜ 3.2360e+03｜ 4.9012e-08
   22｜1.0000e+00｜1.7347e-18｜1.5693e-03｜1.0000e+02｜-1.3972e-06｜ 3.2360e+03｜-1.0732e-07
   23｜1.9531e-03｜8.6736e-19｜1.1761e-03｜1.0000e+01｜-7.6224e-06｜ 3.2360e+03｜ 5.7521e-08
   24｜1.9531e-03｜1.7347e-18｜1.3682e-03｜1.0000e+02｜-1.1035e-06｜ 3.2360e+03｜ 2.4452e-09
   25｜1.9531e-03｜1.7347e-18｜1.4714e-03｜1.0000e+03｜-1.2102e-07｜ 3.2360e+03｜ 2.4920e-10
   26｜1.9531e-03｜1.7347e-18｜1.4754e-03｜1.0000e+04｜-1.2244e-08｜ 3.2360e+03｜ 5.4115e-11
   27｜1.9531e-03｜1.7347e-18｜1.4750e-03｜1.0000e+05｜-1.2258e-09｜ 3.2360e+03｜ 1.1369e-11
   28｜1.5625e-02｜3.4694e-18｜1.4752e-03｜1.0000e+06｜-1.2259e-10｜ 3.2360e+03｜-4.5475e-13
   29｜7.8125e-03｜3.4694e-18｜1.4752e-03｜1.0000e+06｜-1.2259e-10｜ 3.2360e+03｜-1.3642e-12
   30｜3.1250e-02｜3.4694e-18｜1.4752e-03｜1.0000e+07｜-1.2252e-11｜ 3.2360e+03｜-5.0022e-12
   31｜1.9531e-03｜3.4694e-18｜1.4752e-03｜1.0000e+07｜-1.2250e-11｜ 3.2360e+03｜ 6.8212e-12
   32｜3.1250e-02｜4.3368e-18｜1.4752e-03｜1.0000e+08｜-1.2137e-12｜ 3.2360e+03｜-5.4570e-12
   33｜7.8125e-03｜5.2042e-18｜1.4752e-03｜1.0000e+08｜-1.2118e-12｜ 3.2360e+03｜-4.5475e-13
Convergence failure.
Elapsed time:  1.7836387157440186
Results {
  num_iters:    32,
  converged:    false,
  traj. cost:   3.236e+03,
  merit.value:  0.000e+00,
  prim_infeas:  5.204e-18,
  dual_infeas:  1.475e-03,
}
```

1. It seems like the problem formulation is wrong and contains some terms that shouldn't exist for SolverFDDP. Is this normal? Is SolverFDDP simply ignoring these constraints?
2. Despite the warning, the solver still converges (although does not meet the tolerance) and returns a motion that Talos steps in place. How am I supposed to understand the output if the solver fails to converge? Is the result still valid or physically feasible?

## Reproduction steps

I commented line 292 and recovered line 293, so that it's using SolverFDDP instead of SolverProxDDP.
I commented line 298 since this is not a property of SolverFDDP.
Then I run `examples/talos_walk.py` with display option.

### System

- OS: Ubuntu 22.04
- aligator version: [0.6.1]

edantec, 2024-07-29:
As stated by the warning, the FDDP formulation does not handle hard constraints other than dynamics and thus is ignoring them. Nevertheless, the solver FDDP should be able to converge even without constraint. The behaviour you are experiencing is not normal and we will investigate as soon as possible. We are currently proceeding to a major refactoring of the C++ API that should correct these kind of discrepancies.

ManifoldFR, 2024-09-10:
Hello @Cfather, we just saw your paper go out on google scholar and I checked it out. Interesting work, we'll take a closer look, hoping it can give us ideas on making aligator better in the future (although beating Ipopt is quite hard lol)

Would you be interested in contributing some of the examples from [your repository](https://github.com/roahmlab/aligator-roahmlab) back to aligator ? We would be most grateful.

closed 2024-09-10 by ManifoldFR

Cfather, 2024-09-12:
Hi, @ManifoldFR. Thanks for the opportunity! I am definitely interested in contributing to aligator. Could you give me some time to clean up the code that I have? I'm currently preparing for my thesis so I'm afraid my bandwidth is limited. But thank you again for your interest in my work!