Simple-Robotics/aligator #180: conda install aligator pulling pinocchio 2.7.0

Cfather, 2024-08-07:
## Bug description

Hi, I was using conda to install aligator. I did not install pinocchio yet. It seems that conda automatically tried to pull pinocchio 2.7.0 instead of the latest version (3.1.0?).
As a result, some of the example scripts in aligator are not valid since they contain features that are not included in pinocchio 3.1.0 or 3.0.0

## Expected behavior

The dependency of aligator in conda should pull pinocchio 3.1.0 rather than 2.7.0

### System

- OS: [Ubuntu 22.04]
- aligator version: [0.5.0]

ManifoldFR, 2024-08-07:
Hi, 

Have you tried doing 
conda install aligator 'pinocchio>=3'
to force conda to attempt resolution with a more recent Pinocchio?

Cfather, 2024-08-07:
Yep. I am able to run the code if I install pinocchio 3.0.0 first and then install aligator. 
I'm just not sure how conda decides the dependency on pinocchio. Is it automatically decided by conda or you guys can manually setup the desired version of pinocchio on conda forge website?

ManifoldFR, 2024-08-07:
It's both things - the package on conda-forge is compiled against the Pinocchio dependency for the latest version that's pulled when the package is built.
When a new version of the dependency is available we can have conda-forge rebuild the package with the new dependency versions. This means that there's a few possible combinations of Python and Pinocchio and proxsuite-nlp for which a given version of aligator can exist, and you can wrangle Conda to try to get the version you want (typically Conda will try not to change too many packages to get you the latest of everything unless it has to or you ask it to)

Cfather, 2024-08-07:
Thanks for your reply!

closed 2024-08-07 by Cfather