Simple-Robotics/aligator #195: Replace vector of CostPtr in CostStack objects by a map of CostPtr

edantec, 2024-09-16:
When creating a CostStack, the user must keep in mind the index of each Cost component they have defined, in order to be able to modify the Cost object later on. It would be more intuitive to use `std::map<String,CostPtr>` to retrieve each component, instead of using `std::vector<CostPtr>`.

ManifoldFR, 2024-09-17:
We will be using `boost::unordered_map` for now. It is quite fast, faster than `std::unordered_map` in recent versions of Boost (> 1.80)

https://martin.ankerl.com/2022/08/27/hashmap-bench-01/#boost__unordered_map

ManifoldFR, 2024-09-17:
Two other options are robin_hood maps and unordered_dense from the author the benchmark:
- https://github.com/martinus/robin-hood-hashing
- https://github.com/martinus/unordered_dense

PR #199 (2024-09-18): [modelling] Change CostStack backend to an unordered_map type
- Also adds a template getter for the component cost functions
- `addCost()` now has an overload taking a key (either a `std::size_t` or a string)
- `addCost()` returns a reference to the pair of cost function component and weight

Resolves #195
closed 2024-09-18 by ManifoldFR