Simple-Robotics/aligator #209: Cannot set reference for residual function in Python

edantec, 2024-09-20:
## Bug description

In Python, the SetReference function of residuals does not work in some cases, probably because the bindings return a copy of the object somewhere instead of a reference.

## Expected behavior

When holding a CostStack object in Python containing a residual, we should be able to modify the corresponding residual reference.

### Code

```
import numpy as np
import aligator
import pinocchio as pin
import example_robot_data as erd

robot = erd.load("talos")
rmodel = robot.model
RF_id = rmodel.getFrameId("right_sole_link")
nv = rmodel.nv
nu = nv - 6

space = aligator.manifolds.MultibodyPhaseSpace(rmodel)

posref = pin.SE3.Identity()
frame_fn_RF = aligator.FramePlacementResidual(space.ndx, nu, rmodel, posref, RF_id)
rcost = aligator.CostStack(space, nu)
residual = aligator.QuadraticResidualCost(space, frame_fn_RF, np.eye(6))
rcost.addCost(residual)

newpos = pin.SE3.Identity()
newpos.translation[2] = 0.2
print(
    "Translation stored = ",
    str(rcost.components[0][0].residual.getReference().translation),
)
rcost.components[0][0].residual.setReference(newpos)
print(
    "Set reference translation to ",
    str(newpos.translation),
)
print(
    "New translation stored = ",
    str(rcost.components[0][0].residual.getReference().translation),
)
```

## Additional context

Macro `ALIGATOR_EIGENPY_HAS_MAP_SUPPORT` is on.

### System

- OS: Ubuntu 20.04
- aligator version: 0.8.0

edantec, 2024-09-20:
Addendum: this problem has been encountered before and was supposedly solved by this commit: 81dc2332a05cb71bad4d3bf49cb81a2fd25c62dd
The switch from vector to map of components probably caused the problem to arise again.

ManifoldFR, 2024-09-20:
This is my fault, I forgot to overload the base_get_item in the map visitor... 🤦🏼‍♂️

ManifoldFR, 2024-09-23:
Okay, we need to fix this like today

ManifoldFR, 2024-09-23:
I just pushed on devel @edantec

ManifoldFR, 2024-09-23:
Resolved by 36c4639c

commit 09d54374 (2024-09-23): [python] CostStack : use overload_base_get_item_for_map

This commit requires https://github.com/stack-of-tasks/eigenpy/pull/507 to be merged.

Resolves #209
closed 2024-09-23 by ManifoldFR