Simple-Robotics/aligator #218: Reformulation of the parallel Riccati kernel

edantec, 2024-09-30:
The current formulation of the parallel solver does not allow to cycle the solver data easily. As a consequence, we must reallocate the LQR subproblem solver data at each control cycle in MPC context. 

This issue is related to the solver cycling PR in which a first implementation of cycling for parallel solver is proposed (but does not work yet).

abussy-aldebaran, 2025-04-01:
The example `mpc-cycle.cpp` crashes when `linear_solver_choice` is set to `LQSolverChoice::PARALLEL`. Is this related ?

ManifoldFR, 2025-04-01:
It could be if the example has a change of dimensions between two stages.
I'm still working on this in one of the open PRs

abussy-aldebaran, 2025-04-01:
Indeed the number of constraints changes (from 1 to 0).

ManifoldFR, 2025-04-01:
@edantec right now has a branch where the entirety of the parallel solver's data structures are reallocated for this purpose.

I had concluded similarly that reallocation was the simplest solution, a few months back, but that the better option was to use a memory resource or arena for this. this is the approach on that PR i mentioned but it does get kinda involved.