Simple-Robotics/aligator #241: Build without Pinocchio fail

jorisv, 2024-10-30:
## Bug description

When creating a build directory with the following cmake command:

```bash
cmake .. -DBUILD_WITH_PINOCCHIO_SUPPORT=OFF
```

We got the following error message:

```
In file included from /home/jvaillan/dev/inria/prox/aligator/src/modelling/multibody/multibody-wrench-cone.cpp:1:
/home/jvaillan/dev/inria/prox/aligator/include/aligator/modelling/multibody/multibody-wrench-cone.hxx:11:36: error: expected initializer before '<' token
   11 | void MultibodyWrenchConeResidualTpl<Scalar>::evaluate(const ConstVectorRef &x,
      |                                    ^
/home/jvaillan/dev/inria/prox/aligator/include/aligator/modelling/multibody/multibody-wrench-cone.hxx:30:36: error: expected initializer before '<' token
   30 | void MultibodyWrenchConeResidualTpl<Scalar>::computeJacobians(
      |                                    ^
/home/jvaillan/dev/inria/prox/aligator/include/aligator/modelling/multibody/multibody-wrench-cone.hxx:51:1: error: 'MultibodyWrenchConeDataTpl' does not name a type
   51 | MultibodyWrenchConeDataTpl<Scalar>::MultibodyWrenchConeDataTpl(
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~
```

It's because `MultibodyWrenchConeResidualTpl` and `MultibodyWrenchConeDataTpl` are only defined in `multibody-wrench-cone.hpp` when `ALIGATOR_PINOCCHIO_V3` is defined.

Also, there is a lot of `#include <pinocchio/...>`  without any guard. The project can't build without Pinocchio installed.

## Expected behavior

The code should build without Pinocchio installed.

## Reproduction steps

Steps to reproduce the behavior:
```
pixi run build
```

### System

- OS: GNU/Linux, OSX-ARM64, OSX-64
- aligator version: 0.9.0

ManifoldFR, 2024-12-04:
I'll take care of this today ASAP.

ManifoldFR, 2024-12-04:
We have a mix of two things: 
* unguarded includes of Pinocchio (should be okay in files that are *not* compiled/included in compiled files when Pinocchio supported is disabled)
* some files being compiled when they shouldn't be

commit 3576a073 (2024-12-04): Completely handle building without Pinocchio support - even without Pinocchio support in proxsuite-nlp

Resolves #241
closed 2024-12-04 by ManifoldFR