Simple-Robotics/aligator #266: Must I give the analytic form of dforward when defining myself ODEAbstract?

Zionshang, 2025-01-18:
I want to define my own state equation of state $\dot{x} = f(x, u)$ by inheriting `aligator::dynamics::ODEAbstractTpl<double>`.

When I use other optimization methods, such as newton gradient descent in NLP instead of DDP, I use forward difference or center difference to find the gradient.

It is now difficult for me to get the analytic form of the Jacobian of in state equation. Can this differential method be applied to aligator? If so, how to add it to the dForward function.

Thank you very much!!!!!!!!

Zionshang, 2025-01-20:
@ManifoldFR Sorry to bother you, but I'd really like to use aligator

ManifoldFR, 2025-01-20:
I reviewed what's available in the library right now and we have finite-differencing for constraint functions and discrete dynamics: https://github.com/Simple-Robotics/aligator/blob/main/include/aligator/modelling/autodiff/finite-difference.hpp

We also have cost functions. We don't have continuous dynamics (ODEAbsract) yet. We didn't have the time (if you'd like to contribute that, please do!)

You could:
1) just pass your custom dynamics with empty derivatives to an integrator 2) finite-difference the integrator using the relevant class.

Or you could implement `Jx_` and `Ju_` by finite differences yourself in your custom class (might be slow if you're doing it from Python).

Zionshang, 2025-01-20:
@ManifoldFR Thank you very much for your answer!

Merci beaucoup!

I'll try what you said!

If I have time, I'll consider adding automatic differentiation to `dforward` as well!

closed 2025-01-20 by ManifoldFR