Simple-Robotics/aligator #270: The results of the solution did not satisfy the discrete dynamics consistency

Zionshang, 2025-02-03:
## Bug description

I found that the results of the solution did not satisfy the discrete dynamics consistency, where the dynamics is customized by myself.

## Reproduction steps

First, I created a custom dynamics

```cpp
struct MyDynamics : ODEAbstractTpl<double>{}
```

When creating the stage model, I used `aligator::autodiff::DynamicsFiniteDifferenceHelper<double>` since I couldn't provide the analytic solution of 'dforward' function

```cpp
StageModel createStage(const ConstVectorRef &x_ref)
{
    // create rcost
    // ...

    MyDynamics dyn = MyDynamics();
    IntegratorSemiImplEuler discrete_dyn = IntegratorSemiImplEuler(dyn, timestep);
    DynamicsFiniteDifferenceHelper finite_diff_dyn(space, discrete_dyn, finite_diff_step); 
    StageModel stage_mdoel = StageModel(rcost, finite_diff_dyn);
    return stage_mdoel;
}
```

And used this stage model to create an MPC problem

```cpp
void creatMpcProblem(const ConstVectorRef &x0, const std::vector<VectorXd> &x_ref)
{
    std::vector<xyz::polymorphic<StageModel>> stage_models;
    for (size_t i = 0; i < mpc_settings_.horizon; i++)
    {
        StageModel stage = createStage(x_ref[i]);
        stage_models.push_back(std::move(stage));
    }
    // create terminal cost
    // ...

    problem_ = std::make_unique<TrajOptProblem>(x0, std::move(stage_models), term_cost);
}
```
Here are some settings for my solver

```cpp
// solver settings
solver_->rollout_type_ = aligator::RolloutType::LINEAR;
solver_->force_initial_condition_ = true;
solver_->linear_solver_choice = aligator::LQSolverChoice::PARALLEL;
solver_->setNumThreads(8);
solver_->max_iters = 1;  // 100 in the first solve and 1 in warm start
```

In the process of optimization, I found that my results are (partial omission of results)

```cpp
// solver_->results_.xs
results_.xs[0]:  0.0033  0.0016  0.1458 -0.0030 -0.0018 -0.0026  1.0000
results_.xs[1]:  0.0014  0.0042  0.2777 -0.0073  0.0022 -0.0006  1.0000
results_.xs[2]:  0.0002  0.0052  0.2791 -0.0075  0.0018 -0.0005  1.0000

// solver_->results_.us
results_.us[0]: -0.4168  0.2110 -3.3181  0.9045  0.1023 -2.2387 -0.5293  0.2493 -3.0393  0.8475  0.1156 -2.0146
results_.us[1]: -0.5120  0.7154 -3.8740  1.0209  0.2431 -3.1522 -0.5760  0.7406 -3.0201  0.6450  0.2480 -2.3788
results_.us[2]: -0.2678 -0.1970 -3.5340  0.9160 -0.1149 -1.8770 -0.5942 -0.1805 -3.8452  1.0828 -0.1044 -2.0303
```

The expected height of the robot I set is 0.29 (the third element in `xs`). But I noticed that the height of the robot went from 0.1458 to 0.2777 instantly, while the input u was almost the same.

To determine whether my results satisfy the discrete dynamics equations, I did the following verification

```cpp
MyDynamics dyn = MyDynamics();
shared_ptr<ContinuousDynamicsData> dyn_data = dyn.createData();
MyDynamicsData &d = static_cast<MyDynamicsData &>(*dyn_data);
dyn.forward(results_.xs[0], results_.us[0], d);
auto a = dyn_data->xdot_.segment(0, 6);
std::cout << "a: " << a.transpose() << std::endl;

IntegratorSemiImplEuler discrete_dyn = IntegratorSemiImplEuler(dyn, timestep);
IntegratorSemiImplData discrete_data(&discrete_dyn);
discrete_dyn.forward(results_.xs[0], results_.us[0], discrete_data);
auto x_next = discrete_data.xnext_;
std::cout << "x_next: " << x_next.head(7).transpose() << std::endl;
```
the result is

```
a:   0.3841  -0.1602  -2.7822 -17.3082   4.1688  -0.6383
x_next:  0.0061 -0.0005  0.0993 -0.0259  0.0036 -0.0045  0.9996
```

I found `x_next` different from `results_.xs[1]` at all!

I suspect the following points could be causing this issue
1. issues with custom dynamics
2. solver settings
3. only 1 iteration does not guarantee a strict dynamic constraint
4. issues with `aligator::autodiff::DynamicsFiniteDifferenceHelper<double>`

Zionshang, 2025-02-13:
@ManifoldFR, I conducted a simple test in the example [se2-car.cpp](https://github.com/Simple-Robotics/aligator/blob/main/examples/se2-car.cpp). When I set `max_iters = 1`, it resulted in a convergence failure, leading to a situation where dynamic consistency could not be guaranteed.

In my view, for standard DDP algorithms, whether or not convergence is achieved, the optimization results should still maintain dynamic consistency. This is because the dynamics equations are strictly followed during the forward pass.

**Doesn't ProxDDP strictly adhere to dynamic consistency during the forward pass?**

If it doesn't, it's worth noting that ProxDDP is typically iterated only once in the MPC process to ensure real-time performance, making convergence challenging. Consequently, the results may not satisfy the dynamics equations.

**What is the value of an optimized trajectory if it fails to meet even the most fundamental dynamics equations?**

I have a limited understanding of ProxDDP theory, as the literature can be quite complex. If I've misunderstood anything, I appreciate your patience. I would be grateful for any clarification you can provide regarding my question!

ManifoldFR, 2025-02-13:
Hello @Zionshang !
Sorry for not getting back to you sooner. Your second message made your problem clearer.

ProxDDP is a multiple-shooting-type DDP solver, so it can ingest infeasible state-control initial guesses, but is only guaranteed to get a dynamically feasible solution at convergence or after enough iterations. The speed at which it gets to a feasible solution is controlled by the penalty term `mu`, so users should tune the _initial_ penalty `mu_init`, and prefer using the nonlinear rollout strategy.

When using ProxDDP for MPC with a limited iteration budget, one should 1) set `mu_init` to a low value (whether a linear or nonlinear rollout is used) and 2) (optional) the first MPC cycle should have a feasible initial guess, which can be obtained by warming up the solver with a few offline iterations on the first cycle's OCP first.

I hope this has answered your question!

Zionshang, 2025-02-15:
I apologize for my previous shallow understanding of DDP. Thank you so much for your response! It completely resolved my issue!

closed 2025-02-16 by Zionshang