Simple-Robotics/aligator #275: Aligator failed to import from conda

fabinsch, 2025-03-17:
Installing `aligator` in a fresh conda environment results into the following problem on my machine

```
conda create --name aligator python=3.12 --y
conda activate aligator
conda install aligator -yy

(aligator) ➜  ~ python -c "import aligator"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/__init__.py", line 5, in <module>
    from .pyaligator import *
ImportError: libproxsuite-nlp.so.0.10.0: cannot open shared object file: No such file or directory
(aligator) ➜  ~ python -c "import proxsuite_nlp"
(aligator) ➜  ~ ldd ~/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/pyaligator.cpython-312-x86_64-linux-gnu.so 
	linux-vdso.so.1 (0x00007ffe42fb7000)
	libeigenpy.so => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libeigenpy.so (0x00007fd37eab4000)
	libaligator_croc_compat.so.0.10.0 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libaligator_croc_compat.so.0.10.0 (0x00007fd37ea9b000)
	libboost_python312.so.1.86.0 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libboost_python312.so.1.86.0 (0x00007fd37ea59000)
	libaligator.so.0.10.0 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libaligator.so.0.10.0 (0x00007fd37e81d000)
	libaligator_gar.so.0.10.0 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libaligator_gar.so.0.10.0 (0x00007fd37e77a000)
	libproxsuite-nlp.so.0.10.0 => not found
	libfmt.so.11 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libfmt.so.11 (0x00007fd37e73a000)
	libgomp.so.1 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libgomp.so.1 (0x00007fd37e6fb000)
	libpinocchio_default.so.3.3.1 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libpinocchio_default.so.3.3.1 (0x00007fd37e211000)
	libstdc++.so.6 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libstdc++.so.6 (0x00007fd37e02d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd37df46000)
	libgcc_s.so.1 => /home/fschramm/mambaforge/envs/aligator/lib/python3.12/site-packages/aligator/../../../libgcc_s.so.1 (0x00007fd37df27000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd37dcfe000)
	libproxsuite-nlp.so.0.10.0 => not found
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fd37dcf7000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fd37dcf2000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd37dced000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fd37dce8000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fd38005e000)
	libproxsuite-nlp.so.0.10.0 => not found
(aligator) ➜  ~ 
```
Note that `proxsuite-nlp` got installed and it is possible to import, just the linking from `alligator`  seems to be broken.

However, installing the dependencies and building from source (on main) works without problems. 

Do you know what could be wrong?

ManifoldFR, 2025-03-17:
Maybe we need to perform a rebuild on conda...

ManifoldFR, 2025-03-17:
Related: https://github.com/conda-forge/aligator-feedstock/pull/39

ManifoldFR, 2025-03-17:
I also released v0.11.0: #276 
Conda-forge should autoupdate soon

jorisv, 2025-03-17:
Two issues:
- conda-forge run_exports (ABI compatibility) is set to `x.x` but we don't provide ABI compatibility
- aligator link against proxsuite.so.x.y.z, so if x.y.z change, linker will fail

First will be fixed here: https://github.com/conda-forge/proxsuite-nlp-feedstock/pull/45

There is no fix for the second one, but I will check where is done in our CMakeLists.txt.

closed 2025-09-29 by jcarpent