Simple-Robotics/aligator #290: Possible documentation bug in `StateErrorResidual` – sign mismatch in residual computation

vagtsiats, 2025-04-24:
Hi, thanks for your work on `aligator`!

While trying to use `StateErrorResidual` to add joint position limits, I noticed a potential inconsistency between the documentation and the actual behavior of the code.

The [documentation](https://simple-robotics.github.io/aligator/structaligator_1_1detail_1_1StateOrControlErrorResidual.html#details) states that the residual is computed as:
r(z) = z ⊖ z_target
However, in my experiments, it seems the function actually computes:
r(z) = z_target ⊖ z

Here’s a minimal test to illustrate the behavior:
```import unittest
import numpy as np
import pinocchio as pin

import example_robot_data as erd

import aligator
from aligator import manifolds, dynamics, constraints

talos = erd.load("talos")

model = talos.model
data = model.createData()

space = manifolds.MultibodyPhaseSpace(model)

q0 = model.referenceConfigurations["half_sitting"]
v0 = np.zeros(model.nv)
x0 = np.concatenate([q0, v0])

nq = model.nq
nv = model.nv
nu = model.nv - 6

state_fn = aligator.StateErrorResidual(space, nu, space.neutral())

# Check joint position part
joint_pos_fn = state_fn[6:nv]
data = joint_pos_fn.createData()

u0 = np.zeros(model.nv - 6)
joint_pos_fn.evaluate(x0, u0, data)

print(data.value)
print(x0[7:nq])
print(np.allclose(-data.value, x0[7:nq]))
```

This prints True, suggesting the sign is inverted from what the docs say.

It’s also unclear to me if I’m using the API correctly for this purpose. I couldn’t find any examples using StateErrorResidual or another method for enforcing joint position limits directly. If there’s a better or recommended way to do this, I’d really appreciate any pointers.

vagtsiats, 2025-04-24:
Yep, that's what I am doing, and the slice operators are indeed pretty useful. However, since I am passing the state error object to the constraint definition:
`stage_model.addConstraint(joint_pos_fn, jpos_box_cstr)`
and since the operation is inverted, the constraint should be defined as:
`jpos_box_cstr = constraints.BoxConstraint(-rmodel.upperPositionLimit[7:], -rmodel.lowerPositionLimit[7:])`
instead of:
`jpos_box_cstr = constraints.BoxConstraint(rmodel.lowerPositionLimit[7:], rmodel.upperPositionLimit[7:])`

This caused some confusion on my end due to the combination of the inverted operation and the constraint definition, so it took me a while to figure out what was going wrong 😅

The same applies for `joint_vel_fn = state_fn[nv:]`, but since the bounds are `[-v_max, v_max]`, it doesn’t cause any issues.

Maybe the documentation could stay as is, but it would be great if the difference operation could be inverted in the C++ code to maintain consistency across the API.

Thanks again for your help!

ManifoldFR, 2025-04-26:
Hum, you're absolutely right, it _should_ be inverted for consistency. Initial version was definitely a mistake.

I'll open up a PR for it, would you take a look at it to ensure I also fix the documentation and properly update the examples and tests?

ManifoldFR, 2025-04-26:
Fixed by #292.

closed 2025-04-26 by ManifoldFR

ManifoldFR, 2025-05-05:
@edantec You should check out this issue. The change in #292 released as part of 0.13 has "fixed" the way (box/limit) constraints should be added to state variables.

Check @vagtsiats's comment: https://github.com/Simple-Robotics/aligator/issues/290#issuecomment-2827861094

PR #292 (2025-04-26): Reverse sign of StateErrorResidual
Resolves #290.

@vagtsiats this should address your comments.

**EDIT** I also ended up fixing the darn CI and reorganize some of the Python tests